USU-Robosub/Poseidon #14: Ruby does not show C++ exceptions

Jesse-V, 2015-01-27:
The C++ API classes are designed to throw exceptions when something goes wrong, such as passing an invalid argument (-1<x>1) to a method, or creating a second instance of ThrustController. Ruby shows "unknown C++ exception" but doesn't show the exception text. This makes debugging Ruby-side very difficult.

TekuConcept, 2015-02-03:
Common C++ exceptions from <stdexcept> like "invalid_argument" are wrapped automatically. However, specially defined or extended exceptions need to wrapped the same way methods are.

Now since code isn't perfect, some errors like the ones observed, can occure out of the blue, and these need to be manually tracked down to the source and quarantined.

TekuConcept, 2015-02-03:
Ommited text: "...from (stdexcept) like..."

closed 2015-02-03 by TekuConcept