USU-Robosub/Poseidon #26: IMU Controller

nfcopier, 2016-01-08:
We need to be able to receive data from the IMU.

- [ ] IMU Controller is initialized with a connection to the Node app.
- [ ] IMU Controller receives data from the drivers.
- [ ] Node app has interface to the IMU Controller.

The I feel like the best implementation is to have the Node interface constantly pulling for data and caching the most recent data set.

*A note on acronyms in code: The generally convention for acronyms in code is to capitalize only the first letter IE: `ImuController` vs `IMUController`.

TekuConcept, 2016-01-08:
IMU is an acronym in and of its self. It stands for Inertial Measurement Unit; Thus I capitalized it as follows: "IMUController" (In addition, it also looks nicer)

nfcopier, 2016-01-08:
I know it's an acronym. That's why I put that note. I would disagree that `IMUController` looks nicer. I find the lack of clear separation of words makes it difficult to read. This is why most linters discourage that type of capitalization.

TekuConcept, 2016-02-01:
The IMU driver code that uses I2C needs to be implemented on the Arduino while the serial interface needs to be done on the Bone. The existing functions such as "sealevelPresure" also should be implemented on the Arduino in order to preserve efficiency with data transfers.

butterfieldj, 2016-02-01:
Yep, I will get to work on this hopefully tomorrow. Is anyone planning on
being in the lab tomorrow? I still don't have a key card.

TekuConcept, 2016-02-01:
I may be there along with Nathan, and maybe Kevin? Tomorrow we also have the regular Tuesday meeting. (Don't worry about the files: BMP085, HMC5883L, and MPU6050, those are going on the Arduino - I should have them tested by the end of the day)

butterfieldj, 2016-02-01:
Ok, what time are you going to be there?

TekuConcept, 2016-02-01:
Maybe before 12pm, but guaranteed after 4:15pm

butterfieldj, 2016-02-01:
Sounds good, I'll see you guys there.

TekuConcept, 2016-04-13:
Resolved by #93.

closed 2016-04-13 by TekuConcept