USU-Robosub/Poseidon #45: Arduino Hangs on Bad Input

TekuConcept, 2016-02-04:
When an index value outside the bounds of the controller array is passed to the Arduino, the microcontroller becomes unresponsive and requires a physical reset. Should we add a watchdog timer or just go simple and catch all bad input?

nfcopier, 2016-02-04:
It would be trivial to catch out of bounds input.
Adding the line `if(controllerNumber > controllers.size) return;` right after reading in the controllerNumber from Serial ought to do it.

I am wondering whether there might be other types of input errors we're missing, though.

TekuConcept, 2016-02-04:
I can add that line then to my branch before sending a pull request. I will also include the LightController in the code as well.

(The IMU will need to be implemented on the bone unfortunately. I tryied all I could to get I2C properly working on the Due with no success - and we don't have time to write out the low level Sam3XE instructions for that fix. I still have the working files from the old code)

TekuConcept, 2016-02-06:
Closed by #46

closed 2016-02-06 by TekuConcept