USU-Robosub/Poseidon #132: Arduino Reads Garbage On Initial Startup

TekuConcept, 2016-09-07:
When the Arduino's Serial is connected to for the first time, garbage data is given via `Serial.read()`.
What I discovered is that most times, the 'value' of the garbage is consistent but the amount of garbage is random. The value of the garbage is unique to both the baud rate and the Arduino device. For the Arduino Due with a baud of 115200, the garbage value is '0'.

Note: `Arduino.available()` will not resolve this issue.

TekuConcept, 2016-10-03:
Discussing this with @kburgon, sending the char "r" or "R" would be easier to manage instead of the full string "ready".

nfcopier, 2016-10-04:
Of course, complete words are better when possible. But this is one place where it makes sense to abbreviate. Sounds like a good idea to me.

PR #170 (2016-11-09): Added clear buffer
Hey can you look at this. Hopefully I did the pull request right. @nfcopier 

Fixes #132.
closed 2016-11-09 by TekuConcept