USU-Robosub/Poseidon #144: DVL Headers

nfcopier, 2016-09-14:
## Summary
The header files for the DVL should be located or reverse-engineered. Some useful documentation should be created.

## Acceptance Criteria

## Non-functional Criteria

TekuConcept, 2016-09-14:
Just to clarify: You will need to write your own software drivers for the Teledyne Doppler Velocity Log.
There are no pre-existing drivers or linux-based software of any kind provided by Teledyne according to their website: rdinstruments.com

They, however, do provide a detailed explanation of packet information that is sent and received serially. (You will need to create another tty serial interface to the DVL device.)

TekuConcept, 2017-03-19:
DVL code has been written for this device in the Drivers Repository. This issue may now be closed.

closed 2017-10-04 by madfrog54321