USU-Robosub/Poseidon #162: Create an Continuous Integration Server

nfcopier, 2016-09-15:
## Summary

Set up a server to compile and test our code automatically.

## Acceptance Criteria

## Non-functional Criteria

[other non-functional criteria]

nfcopier, 2016-09-21:
It sounds like TravisCI has built-in [Slack integration](https://blog.travis-ci.com/2014-03-13-slack-notifications).

nfcopier, 2016-09-21:
It is also free for open-source use.

hvpham, 2016-09-21:
Cool. I will have a look.

madfrog54321, 2016-09-21:
https://travis-ci.org/ is an option for our CI server.

nfcopier, 2016-09-22:
I added compiling the Arduino sketch as an Acceptance Criteria. Thinking about it, it might take quite a bit of work to compile sketches on the CI server. Should that criteria be moved to a separate issue to be done later?

hvpham, 2016-09-28:
I have look at the Travis CI, it seem to work in our case. I have work with Jenkins CI, which is quite good. I am planning to try both of them to see which one fit better to our need. Now all i need is the server to test them on.