USU-Robosub/Poseidon #164: Upload Video Footage to NAS

nfcopier, 2016-09-20:
## Summary

Our video footage is scattered about various devices. This footage should all be located and uploaded to the NAS. This is blocking #149.

## Acceptance Criteria

- [ ] Both GoPros have had their footage uploaded to the NAS
- [ ] Both All-In-One computers should be checked for footage and have that uploaded to the NAS
- [ ] Directories on the NAS should be well organized and named so that the footage can easily be located

nfcopier, 2016-09-26:
@TekuConcept, has this been done? Can I close this issue?

TekuConcept, 2016-09-26:
Yes - not all of it but most definitely the one that contains all footage one could possibly need. (Including some buoy footage)

TekuConcept, 2016-10-03:
Content can be found on our NAS under
> - "/Downloads/Video/GoPro June Pool Time"
> - "/Downloads/Video/2016 Competition"

closed 2016-10-03 by TekuConcept