USU-Robosub/Poseidon #165: Assess Algorithms for Navigation

nfcopier, 2016-09-21:
## Summary

Our current navigation simply corrects thrust based on camera input.  We need to research more robust algorithms. A few examples that have been suggested are mapping and a neural net.

## Acceptance Criteria

- [ ] Algorithms are assessed with pros and cons
- [ ] Algorithms are presented to the team for input
- [ ] A single algorithm has been selected
- [ ] A game plan is created to convert from our present work to the new algorithm, incrementally

## Non-functional Criteria

An ideal algorithm will probably be an fusion of several different algorithms. So we should explore several different algorithms to get as many ideas as possible.

TekuConcept, 2016-09-22:
The sensors available on the IMU are as follows:
> HMC5883L (Compass): [Datasheet](https://cdn-shop.adafruit.com/datasheets/HMC5883L_3-Axis_Digital_Compass_IC.pdf)
> BMP085 (Pressure & Temperature): [Datasheet](https://www.sparkfun.com/datasheets/Components/General/BST-BMP085-DS000-05.pdf)
> MPU6050 (Accelerometer, Gyroscope, & Temperature): [Datasheet](https://www.invensense.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf)

The already-working source for how to use each of these sensors in C++ can be found in the [Drivers Repository](https://github.com/USU-Robosub/Drivers)

madfrog54321, 2016-09-28:
MonoSlam may be used for image localization. However the absence of any unique features in the water may cause camera localization to be non usable.
https://www.youtube.com/watch?v=mimAWVm-0qA

madfrog54321, 2016-09-28:
For best performance here is the recommended SLAM (Simultaneous localization and mapping) setup:

Orientation:
 - Compass (Low accuracy, long term)
 - Gyroscope (High accuracy, short term)
 - Accelerometer (High accuracy, short term)

Depth:
 - Pressure (Medium accuracy, long term)
 - Accelerometer (High accuracy, short term)

Position:
 - Camera (Low accuracy, long term)
 - Accelerometer (High accuracy, short term)

The only problem is with position. To negate the drift from the accelerometer after integrating twice, a monoSlam camera localization is needed to provide a reference. Using just the accelerometer I can only retain accuracy for a maximum of 1 minute. After the time, data from the accelerometer will be unusable. The only other way of fixing the drift problem is to introduce a new sensor capable of giving low accuracy, but long term position information. An example is GPS, but GPS does not work under water.

If a monoSlam camera localization system cannot be implemented, we will have to go without an internal map of the environment. Orientation and depth can still be used, but position will have to be replaced by a in-the-moment algorithm. An example is the current state machine implementation. It relies on a PID style control system. Slowly correcting any error in the sub's position relative to the recognized gate. The problem with a in-the-moment localization algorithm is its inability to deal with loosing the gate or other target. The sub only knows its position relative to a in-view object.

The team will need to decide wither implementing a monoSlam camera localization system is in the scope of the time allowed for fixing the sub's navigation system. If there is time and it works, a monoSlam camera localization system will give our sub a huge advantage over other subs. The system will allow our sub to know its position in 3D space and plan navigation objectives. This may allow us to use an internal map of the field negating the need to search for objectives. Additionally, if this is developed, our team will be on the cutting edge of mapping and localization systems.

The team needs to decide how much time can be spent in the attempt of implementing the full real time mapping system. I recommend that the orientation and depth systems are actively implemented. In addition, the state machine should be improved with the orientation and depth systems allowing for a more accurate navigation. At the same time the monoSlam camera localization system should be tested in the under water environment to evaluate its potential to anchor the position system. I believe this will allow for the best use of the team's time and resources.

TekuConcept, 2016-09-28:
With a Doppler Velocity Log, teams have already been able to set field points with fairly high accuracy.

Without the DVL, maybe nature will have a few helpful ideas?
> Camera (as the sense of sight) would be used to recognize immediate local space and objects within that space.
> Compass can be used to orient the sub with general precision - doesn't need to be perfect - like feeling the sun on your face in a field when your eyes are closed and knowing which general direction the sun is located.
> Accelerator and Gyroscope for immediate orientation - like blindfolding a person and telling them to walk in a straight line. (Accelerator does have error but will not drift like the gyroscope in terms of tilt.)
> Pressure and Temperature (as the sense of touch) to get a feel for the environment. (also doesn't need to be perfect)
> Hydrophones (as the sense of sound) - adding to the compass to get a further sense of direction.

madfrog54321, 2016-09-28:
A new project provides a faster and more accurate localization using imaging. https://github.com/uzh-rpg/rpg_svo A video demonstrating the software is provided. http://youtu.be/2YnIMfw6bJY

closed 2016-10-29 by madfrog54321