USU-Robosub/Poseidon #177: Extended math library for C++

madfrog54321, 2016-11-08:
## Summary

Because sensor fusion can take a large amount of processing power and requires precise timing, the sensor fusion code should be implemented in the C++ module. To allow the development of the simplified sensor interfaces in the C++ module a extended math library needs to be added to the C++ module. 

## Acceptance Criteria

- [ ] Choose math library.
- [ ] Chosen math library needs to have at least matrix, vector, and quaternion support. 
- [ ] The math library has been added to the C++ module in the repository.

madfrog54321, 2016-11-08:
I have found [Eigen](http://eigen.tuxfamily.org/index.php?title=Main_Page) as a possible math library.

TekuConcept, 2016-11-08:
Depending on your requirements, I could give you all 3 of those classes (already written) - trying to reduce the footprint and dependency requirements.

madfrog54321, 2017-01-06:
I added all the code for Vector3 and Quaternion in commit [e05f350](https://github.com/USU-Robosub/Poseidon/commit/e05f3504c3389170fddb19ba7506757f5a7b8359).

nfcopier, 2017-01-06:
Issues should not be closed until the relevant code has been merged into develop.

closed 2017-10-04 by madfrog54321