USU-Robosub/Poseidon #178: Implementing IPoseProvider in C++

madfrog54321, 2016-11-16:
## Summary

To provide the brain with the position and orientation of the sub, a pose provider is needed. To allow  expansion of the sensor fusion algorithm, a IPoseProvider class needs to be created. At a later time a full sensor fusion or map algorithm will be implemented.

## Acceptance Criteria

When [external actor] does [action], [module]
- [ ] When getPosition is called it should return a Vector3 of the sub's offset from its start location. 
- [ ] When getOrientation is called it should return a Quaternion of the sub's offset from level and start direction.

## Non-functional Criteria

- [ ] The pose provider should be used by the sub's command dispatcher and assembler.

madfrog54321, 2017-01-06:
IPoseProvider has been implemented in commit [e05f350](https://github.com/USU-Robosub/Poseidon/commit/e05f3504c3389170fddb19ba7506757f5a7b8359)

nfcopier, 2017-01-06:
Issues should not be closed until the relevant fix is on develop.

closed 2017-10-04 by madfrog54321